WU-BIMAC/MicroMetaApp-React #304: [BUG] MANAGE SETTINGS --> SamplePositioning remove one prev added element remaining became red

strambc, 2021-03-19:
**Describe the bug**
After additing several elements in SamplePositioningSettings attempted to remove one.
All remaining became red.
No errror

**To Reproduce**
Steps to reproduce the behavior:
1. Go to SamplePositioningSettings
2. Add Elements
3. Enter value in Settings (optional) 
4. Confirm
5. Enter again
6. Attempt to remove one item
7. Remaing items are red

**Expected behavior**
Remaining items should remain green

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Mac
 - Version [e.g. 22]: 0.40

related to #303

strambc, 2021-03-19:
Confirmed
Clicked again 
Now all elements are gone

[image]

strambc, 2021-03-19:
Attempted to add elements again, will only allow to add one. Cannot add more

[image]

strambc, 2021-03-19:
Exited without saving
Entered again
Elements are back in their place and it is possible to delete one with all the left ones remaining green

[image]

strambc, 2021-03-19:
Value that had been entered before #303 not present

strambc, 2021-03-19:
Strange red dot behavior starts again after confirming and getting back in.

Akramhar, 2021-03-21:
should be fixed in 0.41

closed 2021-03-22 by strambc