WU-BIMAC/MicroMetaApp-React #310: [BUG] MANAGE SETTINGS --> Imported values from image file header are counted as missing upon confirm

strambc, 2021-03-22:
**Describe the bug**
When loading settings values from the image file header, values are not recognized as such and appear missing upon confirm

**To Reproduce**
Steps to reproduce the behavior:
1. Manage Settings
2. Load existing Microscope (Lacoste
3. Load existing Image file (Kidney etc.)
4. Values that were loaded are correct but they appear RED when hit confirm.
5. When cut and pasted back in they validate

**Expected behavior**
The values should register as correct and validate

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: v041

Akramhar, 2021-03-22:
41.2 has a tentative solution for this problem

strambc, 2021-03-22:
Now it works for float but it does not work anymore for int

[image]

Akramhar, 2021-03-22:
should be done in 0.42, this was a model issue with PositiveInteger in the xsd model not being parsed correctly ("should be a string", instead of should be a number/integer)

strambc, 2021-03-23:
NOT WORKING

closed 2021-03-24 by strambc