WU-BIMAC/MicroMetaApp-React #311: v041 CHECK image loading: Tested all Use-Cases to see if images load

strambc, 2021-03-22:
- [ ] Boehm
- [ ] Brown
- [ ] Chambers
- [ ] Coleman
- [ ] Guilbert
- [ ] Faklaris
- [ ] Itano
- [ ] Lacoste
- [ ] Laude
- [ ] Nelson
- [ ] Nitschke
- [ ] Montero
- [ ] Pimentel
- [ ] Strambio

strambc, 2021-03-22:
GUILBERT ERROR
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.41.0-b1-0/41_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88 Warning: Received the string `false` for the boolean attribute `checked`. The browser will interpret it as a truthy value. Did you mean checked={false}?
    in input (created by FormCheckInput)
    in FormCheckInput (created by FormCheck)
    in div (created by FormCheck)
    in FormCheck (created by CheckboxWidget)
    in div (created by FormGroup)
    in FormGroup (created by CheckboxWidget)
    in CheckboxWidget
    in Unknown (created by BooleanField)
    in BooleanField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by ObjectField)
    in div (created by Col)
    in Col (created by ObjectFieldTemplate)
    in div (created by Row)
    in Row (created by ObjectFieldTemplate)
    in div (created by Container)
    in Container (created by ObjectFieldTemplate)
    in ObjectFieldTemplate (created by ObjectField)
    in ObjectField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by Form)
    in form (created by Form)
    in Form (created by ForwardRef)
    in ForwardRef
    in div (created by TabPanel)
    in TabPanel (created by MultiTabFormWithHeaderV3)
    in div (created by UncontrolledTabs)
    in UncontrolledTabs (created by Tabs)
    in Tabs (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanel)
    in TabPanel (created by MultiTabFormWithHeaderV3)
    in div (created by UncontrolledTabs)
    in UncontrolledTabs (created by Tabs)
    in Tabs (created by MultiTabFormWithHeaderV3)
    in div (created by MultiTabFormWithHeaderV3)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by MultiTabFormWithHeaderV3)
    in MultiTabFormWithHeaderV3 (created by Footer)
    in Footer (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent
printWarning @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.41.0-b1-0/41_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88
child_process.js:630 Uncaught (in promise) Error: Command failed: java -jar /Users/strambc/MicroscopyMetadataTool/scripts/4DNImageMetadataReader-0.0.1.jar "/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/L_Guilbert/RawData (Jubrail et al. 2020)/E_JJ_Exp17_04052017_DB_CR3_t1_RV16_cv49_02_ROR_SCL.nd"
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at java.lang.StringCoding$StringDecoder.decode(StringCoding.java:149)
	at java.lang.StringCoding.decode(StringCoding.java:193)
	at java.lang.String.<init>(String.java:426)
	at java.lang.String.<init>(String.java:491)
	at loci.common.DataTools.readFile(DataTools.java:91)
	at loci.formats.in.MetamorphReader.initFile(MetamorphReader.java:473)
	at loci.formats.FormatReader.setId(FormatReader.java:1389)
	at loci.formats.ImageReader.setId(ImageReader.java:843)
	at edu.umassmed.imageMetadataReader.ImageMetadataReader.main(ImageMetadataReader.java:514)

    at checkExecSyncError (child_process.js:630:11)
    at execSync (child_process.js:666:15)
    at electron/js2c/asar.js:727:23
    at MicroscopyMetadataToolComponent.onLoadMetadata (webpack://MicroscopyMetadataToolStandAlone/./src/app.js?:502:22)
    at eval (webpack://MicroscopyMetadataToolStandAlone/./node_modules/4dn-microscopy-metadata-tool/es/components/imageLoader.js?:154:22)
    at Array.forEach (<anonymous>)
    at ImageLoader.dropzoneDropAccepted (webpack://MicroscopyMetadataToolStandAlone/./node_modules/4dn-microscopy-metadata-tool/es/components/imageLoader.js?:151:21)
    at eval (webpack://MicroscopyMetadataToolStandAlone/./node_modules/react-dropzone/dist/es/index.js?:698:11)

Akramhar, 2021-03-22:
guilbert E_JJ_Exp17_04052017_DB_CR3_t1_RV16_cv49_02_ROR_SCL.nd doesnt work even with ImageJ bio-formats importer so not sure what's going on with that image

closed 2021-03-22 by strambc