WU-BIMAC/MicroMetaApp-React #315: [BUG] MANAGE INSTRUMENT --> Lots of warnings

strambc, 2021-03-23:
**Describe the bug**
Several warnings appear in developer tools in v042 upon launch

---
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:24994 Download the React DevTools for a better development experience: https://fb.me/react-devtoolsYou might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
electron/js2c/renderer_init.js:83 Electron Deprecation Warning The 'remote' module is deprecated and will be disabled by default in a future version of Electron. To ensure a smooth upgrade and silence this warning, specify {enableRemoteModule: true} in the WebPreferences for this window.
(anonymous) @ electron/js2c/renderer_init.js:83
DevTools failed to load SourceMap: Could not load content for file:///Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata%20standard%20+%20software_model%20development/4DN-OME_Micro-Meta-App_devel/Micro-Meta%20App%20Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/public/assets/css/bootstrap.min.css.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/@rjsf/bootstrap-4/dist/bootstrap-4.esm.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/uri-js/dist/es5/uri.all.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/index.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file-selector.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/html2canvas/dist/html2canvas.js.map:

strambc, 2021-03-23:
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:24994 Download the React DevTools for a better development experience: https://fb.me/react-devtoolsYou might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
electron/js2c/renderer_init.js:83 Electron Deprecation Warning The 'remote' module is deprecated and will be disabled by default in a future version of Electron. To ensure a smooth upgrade and silence this warning, specify {enableRemoteModule: true} in the WebPreferences for this window.
(anonymous) @ electron/js2c/renderer_init.js:83
DevTools failed to load SourceMap: Could not load content for file:///Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata%20standard%20+%20software_model%20development/4DN-OME_Micro-Meta-App_devel/Micro-Meta%20App%20Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/public/assets/css/bootstrap.min.css.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/@rjsf/bootstrap-4/dist/bootstrap-4.esm.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/uri-js/dist/es5/uri.all.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/index.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file-selector.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/html2canvas/dist/html2canvas.js.map:

Akramhar, 2021-03-23:
I removed the warning all the warning that were easy to remove in the 0.42.1, the others are going to need more work in future

closed 2021-03-23 by strambc