WU-BIMAC/MicroMetaApp-React #316: [BUG] MANAGE INSTRUMENT/--> Menus appear at the top left corner of the app

strambc, 2021-03-23:
**Describe the bug**
After the application has been running for a while menus appear at the top left corner of the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Happened with Tier selector and Save button in both Manage Instrument and Settings

**Screenshots**
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS Mac
 - Version v042.0

**Additional context**
Might be related to issue #

strambc, 2021-03-23:
[image]

strambc, 2021-03-23:
@Akramhar  Appears to happen upon clicking again on a selector menu that had been already clicked before.
Appears to happen upon clicking again on the same button

Akramhar, 2021-03-23:
had to roll back a library to avoid this error, should be fixed in 0.42.1
library react-bootstrap 1.5.2 -> rolled back to 1.4.0

closed 2021-03-23 by strambc