WU-BIMAC/MicroMetaApp-React #317: [BUG/QUESTION] MANAGE INSTRUMENT --> Menus are different. Is it due to the new libraries?

strambc, 2021-03-23:
**Describe the issue**
Menus are different and always require scrolling. Is it due to the new libraries?
Is this related to #316 

**To Reproduce**
All menus appear to require scrolling while they did not before.

**OLD BEHAVIOR**
[image]
[image]

**NEW BEHAVIOR**
[image]
[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac
 - Version v042.0

**Additional context**
Might be related to #316

Akramhar, 2021-03-23:
this is how they look on windows, now they look the same on mac too.
I'm guessing its because of updated libraries.

Akramhar, 2021-03-23:
behavior revert to previous due to #316

closed 2021-03-23 by strambc