WU-BIMAC/MicroMetaApp-React #333: [BUG] MANAGE INSTRUMENT/SETTINGS --> After adding a new element to "Add additional element(s)" the buttons shift all the way to the right

strambc, 2021-03-23:
**Describe the bug**
After adding a new element to "Add additional element(s)" the buttons shift all the way to the right

**To Reproduce**
Steps to reproduce the behavior:
1. Click on "Add additional element(s)"
2. Click on an element and add
3. See error: buttons move all the way to the right

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Mac
 - Version: v042.0

**Additional context**
Might be related to #316 and #317

strambc, 2021-03-23:
Uncaught TypeError: Cannot read property 'modelSettings' of null
    at ChannelCanvas_V2.onAddAdditionalConfirm (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/channelCanvas_V2.js?:502)
    at MultiTabFormWithHeaderV3.processData (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/multiTabFormWithHeaderV3.js?:545)
    at MultiTabFormWithHeaderV3.onSubmit (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/multiTabFormWithHeaderV3.js?:461)
    at Form.eval (webpack://MicroMetaAppElectron/./node_modules/@rjsf/core/dist/es/components/Form.js?:259)
    at callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:12490)
    at commitUpdateQueue (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:12511)
    at commitLifeCycles (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19858)
    at commitLayoutEffects (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22803)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527 The above error occurred in the <Form> component:
    in Form (created by ForwardRef)
    in ForwardRef
    in div (created by TabPanel)
    in TabPanel (created by MultiTabFormWithHeaderV3)
    in div (created by UncontrolledTabs)
    in UncontrolledTabs (created by Tabs)
    in Tabs (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanel)
    in TabPanel (created by MultiTabFormWithHeaderV3)
    in div (created by UncontrolledTabs)
    in UncontrolledTabs (created by Tabs)
    in Tabs (created by MultiTabFormWithHeaderV3)
    in div (created by MultiTabFormWithHeaderV3)
    in MultiTabFormWithHeaderV3 (created by ChannelCanvas_V2)
    in div (created by ChannelCanvas_V2)
    in div (created by ChannelCanvas_V2)
    in div (created by ChannelCanvas_V2)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by ChannelCanvas_V2)
    in ChannelCanvas_V2 (created by ChannelView)
    in ChannelView (created by SettingMainView)
    in SettingMainView (created by MicroMetaAppReact)
    in div (created by MicroMetaAppReactContainer)
    in MicroMetaAppReactContainer (created by MicroMetaAppReact)
    in MicroMetaAppReact (created by MicroMetaAppElectronComponent)
    in MicroMetaAppElectronComponent

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
logCapturedError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11102 Uncaught TypeError: Cannot read property 'modelSettings' of null
    at ChannelCanvas_V2.onAddAdditionalConfirm (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/channelCanvas_V2.js?:502)
[... cut ...]
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/html2canvas/dist/html2canvas.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata%20standard%20+%20software_model%20development/4DN-OME_Micro-Meta-App_devel/Micro-Meta%20App%20Software/react_beta_releases/beta_0.42.0-b1-0/42_micro-meta-app-electron.app/Contents/Resources/app.asar/public/assets/css/bootstrap.min.css.map:

Akramhar, 2021-03-23:
unable to reproduce, needs more investigation

Akramhar, 2021-03-25:
the problem is graphical: when resize the window to a small dimension the buttons are shifted

closed 2021-05-15 by strambc