WU-BIMAC/MicroMetaApp-React #338: [FATAL ERROR] MANAGE SETTINGS --> Screen goes blank after hitting Edit Channel

strambc, 2021-03-24:
**Describe the bug**
Working on an existing settings file.
Opened second time and skipping image load

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Edit Channels
2. Click on Edit selected
 BLANK SCREEN

strambc, 2021-03-24:
/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:24994 Download the React DevTools for a better development experience: https://fb.me/react-devtoolsYou might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
electron/js2c/renderer_init.js:83 Electron Deprecation Warning The 'remote' module is deprecated and will be disabled by default in a future version of Electron. To ensure a smooth upgrade and silence this warning, specify {enableRemoteModule: true} in the WebPreferences for this window.
(anonymous) @ electron/js2c/renderer_init.js:83
app.js:582 CleaningFiles from /Users/strambc/MicroMetaApp/schemas
app.js:584 CleaningFiles from /Users/strambc/MicroMetaApp/dimensions
app.js:586 CleaningFiles from /Users/strambc/MicroMetaApp/scripts
app.js:588 CleaningFiles from /Users/strambc/MicroMetaApp/scripts/dependency-jars
app.js:592 CopyFiles from /private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/schemas
app.js:597 CopyFiles from /private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/dimensions
app.js:602 CopyFiles from /private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/scripts
app.js:607 CopyFiles from /private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/scripts/dependency-jars
microscopeLoader.js:269 Array(6)
microscopeLoader.js:269 Array(6)
microscopeLoader.js:269 Array(6)
microscopeLoader.js:269 Array(1)
microscopeLoader.js:269 Array(6)
microscopeLoader.js:269 Array(6)
microscopeLoader.js:269 Array(1)
microscopeLoader.js:269 Array(1)
28bundle.js:6 enter
canvas.js:885 Object
11bundle.js:6 enter
canvas.js:885 Object
canvas.js:885 Object
2bundle.js:6 enter
canvas.js:885 Object
44bundle.js:6 enter
canvas.js:885 Object
3bundle.js:6 enter
canvas.js:885 Object
11bundle.js:6 enter
canvas.js:885 Object
canvas.js:885 Object
30bundle.js:6 enter
canvas.js:966 deletedID
canvas.js:967 ca7c5c12-c83f-425b-855a-79a7d364ba93
31bundle.js:6 enter
canvas.js:885 Object
11bundle.js:6 enter
canvas.js:885 Object
13bundle.js:6 enter
canvas.js:885 Object
canvas.js:885 Object
49bundle.js:6 enter
canvas.js:885 Object
40bundle.js:6 enter
canvas.js:885 Object
18bundle.js:6 enter
canvas.js:885 Object
canvas.js:885 Object
35bundle.js:6 enter
canvas.js:885 Object
canvas.js:885 Object
4bundle.js:6 enter
canvas.js:885 Object
12bundle.js:6 enter
canvas.js:885 Object
canvas.js:966 deletedID
canvas.js:967 8b6d0433-5b50-4379-9c3e-50a397e6d09a
12bundle.js:6 enter
canvas.js:885 Object
8bundle.js:6 enter
canvas.js:885 Object
bundle.js:6 enter
canvas.js:885 Object
2bundle.js:6 enter
canvas.js:885 Object
5bundle.js:6 enter
canvas.js:885 Object
3bundle.js:6 enter
[... 158 lines cut ...]
    at _loop (multiTabFormWithHeaderV3.js:71)
    at new MultiTabFormWithHeaderV3 (multiTabFormWithHeaderV3.js:105)
    at constructClassInstance (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:12880)
    at updateClassComponent (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17100)
    at beginWork (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at beginWork$1 (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23203)
    at performUnitOfWork (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22157)
[... cut ...]
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file-selector.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/1DE4149E-71CE-493F-887F-CABD01682D54/d/micro-meta-app-electron.app/Contents/Resources/app.asar/public/assets/css/bootstrap.min.css.map:

strambc, 2021-03-24:
Application had been running continuously for 2-3 hours

Akramhar, 2021-03-25:
linked to #340, should be fixed in 0.44

closed 2021-03-25 by Akramhar