WU-BIMAC/MicroMetaApp-React #340: [BUG] MANAGE INSTRUMENT/SETTINGS --> Error upon Channel confirm

strambc, 2021-03-24:
**Describe the bug**
Confirmed channel without changing any attributes 

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Edit Channels
2. Click on Edit Selected
3. Hit confirm 
4. Error 

**Desktop (please complete the following information):**
 - OS: MAC
 - Version 43

strambc, 2021-03-24:
Uncaught TypeError: Cannot set property 'ComponentSettings' of undefined
    at ChannelCanvas_V2.onConfirm (channelCanvas_V2.js:169)
    at HTMLUnknownElement.callCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:327 Uncaught TypeError: Cannot set property 'ComponentSettings' of undefined
    at ChannelCanvas_V2.onConfirm (channelCanvas_V2.js:169)
    at HTMLUnknownElement.callCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
channelCanvas_V2.js:169 Uncaught TypeError: Cannot set property 'ComponentSettings' of undefined
[... cut ...]
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file-selector.js.map: 
DevTools failed to load SourceMap: Could not load content for webpack://MicroMetaAppElectron/node_modules/file-selector/dist/es5/file.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///private/var/folders/lj/w2ftks1j43l52wh5tw9cy5540000gr/T/AppTranslocation/D006B0E8-68C9-459F-8296-127F8898D3AD/d/43_GitHub_micro-meta-app-electron.app/Contents/Resources/app.asar/public/assets/css/bootstrap.min.css.map:

Akramhar, 2021-03-25:
linked to #338, should be fixed in 0.44

closed 2021-03-25 by Akramhar