WU-BIMAC/MicroMetaApp-React #356: NEW FEATURE --> WHAT TO DO WHEN MIC CONFIGURATION CHANGES

strambc, 2021-04-01:
What should be done when a component of a Microscope is removed and the Settings file with that component is opened and it does not find the expected component?

There has to be a way of associating an ID with a STATE

Use case 1 is a Microscope whose components have to be replaced

1. Microscope1_State 1 (DATE 1)--> this is the 
2. Microscope1_Sate 2  (DATE 2) --> it could be differentiated using an ID and a DATE or simply a new ID

Use case 2 is a Microscope that has multiple configurations at the same time 
For example 

1. Config 1 has Obj 1, 2, 3, and Laser 1 -- > ID1
2. Config 2 has Obj 3, 4, 5, and Laser 2 --> ID2

For documentation purposes 1 and 2 in both cases are TWO DIFFERENT MICROSCOPES

strambc, 2021-04-01:
No matter what the SETTINGS are the most important and central part of the Metadata and they univocally depend on a given MICROSCOPE CONFIGURATION OR STATE.

glynnelson, 2021-04-01:
I would say it would be nice to have both- some microscopes are frequently swapped between two configuratons that don't change, but sometimes there are other configurations that are rare and only for one project or a test set of data, so would be be a temporary state.  Using microscope_config_AND_Date_AND_modificationVerson would be good for this, whereas microscope_config as a name for the multiple configs would work.