WU-BIMAC/MicroMetaApp-React #357: [BUG] MANAGE INSTRUMENT--> v46-1 attempt to open file from v42-1

strambc, 2021-04-01:
**Describe the bug**
Using v46-1 attempting to a file from 42-1

**To Reproduce**
Steps to reproduce the behavior:
1. Launch app
2. Load from repo
3. ERROR 

**Desktop (please complete the following information):**
 - OS: MaOS
 - Version 46-1

**Additional context**

[COLEMAN_single_molecule_tracker_7_2021-03-09.txt](https://github.com/WU-BIMAC/MicroMetaApp-React/files/6246706/COLEMAN_single_molecule_tracker_7_2021-03-09.txt)

strambc, 2021-04-01:
app.js:1021 Uncaught TypeError: Cannot read property 'split' of undefined
    at MicroMetaAppReact.applyPreviousModelVersionModificationToMicroscope (app.js:1021)
    at MicroMetaAppReact.applyPreviousVersionModificationToMicroscope (app.js:811)
    at MicroMetaAppReact.createOrUseMicroscopeFromSelectedFile (app.js:1226)
    at MicroMetaAppReact.createOrUseMicroscope (app.js:1347)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
applyPreviousModelVersionModificationToMicroscope @ app.js:1021
applyPreviousVersionModificationToMicroscope @ app.js:811
createOrUseMicroscopeFromSelectedFile @ app.js:1226
createOrUseMicroscope @ app.js:1347
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389
executeDispatchesInOrder @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414
executeDispatchesAndRelease @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278
executeDispatchesAndReleaseTopLevel @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287
[... cut ...]
discreteUpdates @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/0.46.1-b1-0/46-1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

Akramhar, 2021-04-02:
should be solved in 0.46.2

strambc, 2021-04-02:
Tested with Microscope.json files from versions bask as far as January 2021
Tested both Load from File and Repository

closed 2021-04-02 by strambc