WU-BIMAC/MicroMetaApp-React #367: [BUG] MANAGE INSTRUMENT/SETTINGS --> concise description

JSDLacoste, 2021-04-08:
**After selecting the AS to manage settings, application stays on "Loading file"**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
First I select the microscope I want to work on. Then:
1. Go to 'Manage Settings'
2. Click on 'Load from file', and select the AS file in my Phase 2 dropbox folder.  Note the file opens correctly with Fiji.
3. At this point "Loading file" is greyed out and nothing else ever happen.

**Expected behavior**
I was hoping for the AS to load so that I can proceed with "Manage Settings".

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Mac 10.14 Mojave
 - Browser Chrome
 - Version Version 89.0.4389.114 (Official Build) (x86_64)

**Additional context**
Add any other context about the problem here.

strambc, 2021-04-08:
Dear Judith
I think I have reproduced the error. To double check, could you please do me a favor and check open the Developers Tools on your side?
They are found under the “View” menu.
Once you have opened the "Developers Tools" please copy the RED error and paste it in as a comment at the bottom of the ISSUE.

**If you cannot find them please check the attached short video.**
https://user-images.githubusercontent.com/4050585/114070748-51bb8380-986e-11eb-9f87-50cd0f92855d.mov

strambc, 2021-04-08:
Tested issue and found the following error

Uncaught (in promise) Error: Command failed: java -jar /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/Z_Lacoste/Phase 2/scripts/4DNMicroscopyMetadataReader-0.0.1.jar "/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/Z_Lacoste/Phase 2/Kidney 40X RGB Zstack 2_LacosteAuthor.czi"
Error: Unable to access jarfile /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata

    at checkExecSyncError (child_process.js:630)
    at execSync (child_process.js:666)
    at electron/js2c/asar.js:727
    at MicroMetaAppElectronComponent.onLoadMetadata (app.js:458)
    at eval (imageLoader.js:156)
    at Array.forEach (<anonymous>)
    at ImageLoader.dropzoneDropAccepted (imageLoader.js:153)
    at eval (index.js:699)

strambc, 2021-04-08:
The error could be due to one of two things:

1. Path for HomeFolder is too long
2. Path for HomeFolder contains SPACES

Tested and verified it is the second

[image]

strambc, 2021-04-08:
HAVE TO FIX IN NEXT RELEASE for this ITERATION

Akramhar, 2021-04-08:
the problem is due to spaces in the name of the jar file

JSDLacoste, 2021-04-08:
From the Developer Tools console:
Uncaught (in promise) Error: Command failed: java -jar /Users/judithlacoste/Dropbox/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/Z_Lacoste/Phase 2/scripts/4DNMicroscopyMetadataReader-0.0.1.jar "/Users/judithlacoste/MicroMetaApp/Kidney_40X_RGB_Zstack 2_LacosteAuthor.czi"
Unable to access jarfile /Users/judithlacoste/Dropbox/Micro-Meta

    at checkExecSyncError (child_process.js:630)
    at execSync (child_process.js:666)
    at electron/js2c/asar.js:727
    at MicroMetaAppElectronComponent.onLoadMetadata (app.js:458)
    at eval (imageLoader.js:156)
    at Array.forEach (<anonymous>)
    at ImageLoader.dropzoneDropAccepted (imageLoader.js:153)
    at eval (index.js:699)

JSDLacoste, 2021-04-08:
Uncaught (in promise) Error: Command failed: java -jar /Users/judithlacoste/MicroMetaApp/scripts/4DNMicroscopyMetadataReader-0.0.1.jar "/Users/judithlacoste/Dropbox/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/Z_Lacoste/Phase 2/Kidney 40X RGB Zstack 2_LacosteAuthor.czi"
Exception in thread "main" java.lang.UnsupportedClassVersionError: edu/umassmed/microscopyMetadataReader/MicroscopyMetadataReader : Unsupported major.minor version 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

    at checkExecSyncError (child_process.js:630:11)
    at execSync (child_process.js:666:15)
    at electron/js2c/asar.js:727:23
    at MicroMetaAppElectronComponent.onLoadMetadata (webpack://MicroMetaAppElectron/./src/app.js?:458:22)
    at eval (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/imageLoader.js?:156:22)
    at Array.forEach (<anonymous>)
    at ImageLoader.dropzoneDropAccepted (webpack://MicroMetaAppElectron/./node_modules/micro-meta-app-react/es/components/imageLoader.js?:153:21)
    at eval (webpack://MicroMetaAppElectron/./node_modules/react-dropzone/dist/es/index.js?:699:11)

strambc, 2021-04-08:
> From the Developer Tools console:
> Uncaught (in promise) Error: Command failed: java -jar /Users/judithlacoste/Dropbox/Micro-Meta App publication/Micro-Meta App_CO-AUTHORS MATERIAL/Z_Lacoste/Phase [...]

**THIS ISSUE IS THE ONE WE FOUND AND HAD TO DO WITH THE SPACE in the Jar file path**

Akramhar, 2021-04-08:
the issue with the spaces within the path should be solved in version 1.0.0, same for the issue regarding the java version (see #368)

JSDLacoste, 2021-04-09:
Here is the java version info:
Last login: Sun Apr  4 18:04:25 on console
MacBook-Pro-3:~ judithlacoste$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)
MacBook-Pro-3:~ judithlacoste$

closed 2021-05-15 by strambc