WU-BIMAC/MicroMetaApp-React #369: [BUG] MANAGE INSTRUMENT/SETTINGS --> cannot load images

paulamonterollopis, 2021-04-11:
**Describe the bug**
I am unsure if this is the same bug reported before, but after selecting the Image file to load, it never actually loads. I am unsure if this is the fact that the file is a .nd2 file or something else.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mojave Version 10.14.6
 - Browser Chrome
 - Version Version 89.0.4389.90 (Official Build) (x86_64)

**Additional context**
Add any other context about the problem here.

paulamonterollopis, 2021-04-11:
I tested changing the name of the file, changing the format to .tif and loading it from the desktop instead of dropbox. Still it does not allow me to load any image file.

Akramhar, 2021-04-21:
this was a java version issue, should be solved

closed 2021-04-21 by Akramhar