WU-BIMAC/MicroMetaApp-React #372: [BUG] MANAGE INSTRUMENT/SETTINGS --> concise description

strambc, 2021-04-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88 Warning: Encountered two children with the same key, `div-New Filament`. Keys should be unique so that components maintain their identity across updates. Non-unique keys may cause children to be duplicated and/or omitted — the behavior is unsupported and could change in a future version.
    in div (created by ChannelCanvas_V2)
    in div (created by ChannelCanvas_V2)
    in div (created by ChannelCanvas_V2)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by ChannelCanvas_V2)
    in ChannelCanvas_V2 (created by ChannelView)
    in ChannelView (created by SettingMainView)
    in SettingMainView (created by MicroMetaAppReact)
    in div (created by MicroMetaAppReactContainer)
    in MicroMetaAppReactContainer (created by MicroMetaAppReact)
    in MicroMetaAppReact (created by MicroMetaAppElectronComponent)
    in MicroMetaAppElectronComponent
printWarning @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88
channelCanvas_V2.js:302 Uncaught TypeError: Cannot read property 'title' of undefined
    at ChannelCanvas_V2.onInnerElementDataSave (channelCanvas_V2.js:302)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:327 Uncaught TypeError: Cannot read property 'title' of undefined
    at ChannelCanvas_V2.onInnerElementDataSave (channelCanvas_V2.js:302)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_relea [... cut ...]
    in div (created by ChannelCanvas_V2)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by ChannelCanvas_V2)
    in ChannelCanvas_V2 (created by ChannelView)
    in ChannelView (created by SettingMainView)
    in SettingMainView (created by MicroMetaAppReact)
    in div (created by MicroMetaAppReactContainer)
    in MicroMetaAppReactContainer (created by MicroMetaAppReact)
    in MicroMetaAppReact (created by MicroMetaAppElectronComponent)
    in MicroMetaAppElectronComponent

strambc, 2021-04-19:
channelCanvas_V2.js:302 Uncaught TypeError: Cannot read property 'title' of undefined
    at ChannelCanvas_V2.onInnerElementDataSave (channelCanvas_V2.js:302)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
onInnerElementDataSave @ channelCanvas_V2.js:302
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389
executeDispatchesInOrder @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414
[... cut ...]
discreteUpdates @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

strambc, 2021-04-19:
channelCanvas_V2.js:302 Uncaught TypeError: Cannot read property 'title' of undefined
    at ChannelCanvas_V2.onInnerElementDataSave (channelCanvas_V2.js:302)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
onInnerElementDataSave @ channelCanvas_V2.js:302
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389
executeDispatchesInOrder @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/1.0.0-b1-0/1-0_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414
[... cut ...]
    in div (created by ChannelCanvas_V2)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by ChannelCanvas_V2)
    in ChannelCanvas_V2 (created by ChannelView)
    in ChannelView (created by SettingMainView)
    in SettingMainView (created by MicroMetaAppReact)
    in div (created by MicroMetaAppReactContainer)
    in MicroMetaAppReactContainer (created by MicroMetaAppReact)
    in MicroMetaAppReact (created by MicroMetaAppElectronComponent)
    in MicroMetaAppElectronComponent

strambc, 2021-04-20:
This issue is related to #376

Akramhar, 2021-04-21:
should be fixed in electron in 1.0.0-b1-1

closed 2021-05-15 by strambc