WU-BIMAC/MicroMetaApp-React #385: STRAMBIO/NITSCHKE -- Increase number of slots for MultiLaserEngine from 6 to 8

strambc, 2021-05-14:
E.g.: Strambio -- Camera/ConversionFactor and Camera/ReadNoise should depend on CameraMode

**Element name(s): LightSourceGroup/MultiLaserEngine**

Justification for the addition (brief and compelling):
This suggested change came from a conversation with Roland Nitschke.
Roland mentioned that several MultiLaserEngines have 7 slots.
It was decided that 8 would be a good idea.

Full description of the proposed addition (enough detail to implement):
Modify the MultiLaserEngine icon as well as the Canvas slot for the MultiLaserEngine on the Canvas.

**Current Element name:** LightSource

**Source of change (e.g., #XXX issue with change request):** Roland Nitschke <br/>

THIS IS RELATED TO ISSUE #

strambc, 2022-03-30:
Planned for in release where Confocal Scanner will be released

Akramhar, 2025-03-18:
Images are already done, @strambc please link dropbox folder with images from Shannon

strambc, 2025-09-18:
@Akramhar we need to discuss this.
Changing the Multilaserengine icon requires changing the underlying canvas, which in turn requires adding UP and DOWN Cameras.

[image]