WU-BIMAC/MicroMetaApp-React #393: [4DN-Portal] List of Caterina's comments issues

strambc, 2021-07-15:
- Do we allow users to delete Microscope.JSON from the 4DN-portal?
- - MY OPINION YES
- We have to decide on how to present the data in tabular form when pressing "Expand"
- - RIGHT NOW IT IS NOT USABLE and VERY CONFUSING
- What do we do with the EDIT button? Do changes there reflect on the JSON and on the microscope information as visualized in the App?

strambc, 2021-07-16:
-- Upload Settings.JSON file is missing

andreacosolo, 2021-07-16:
> * Do we allow users to delete Microscope.JSON from the 4DN-portal?
I'd say no, unless there is a clear use case. Data submitters come to the 4DN data portal and want to document their microscope. They might want to make changes, which they can as many times as needed. Why would they delete the microscope? The actual instrument is still in their lab. Submitters should be instructed not to create multiple duplicated objects for the same instrument, unless they actually change a component on the instrument itself.

strambc, 2021-07-27:
This is related to #400 and #399 
Items 4-6 that were listed here have been moved to #400