WU-BIMAC/MicroMetaApp-React #394: [4DN-Portal] Caterina #1: edited values are not saved

strambc, 2021-07-21:
**Describe the bug**
When attempting to change the metadata values the new values are not saved.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch M2A
2. Click on "Edit Microscope"
2. Click on the "ManufacturerSpecs" tab
3. Edit "m" to "m2" etc.
4. Click Confirm
5. Click on "Edit Microscope" (with or without hitting "Save").
6. Old value is found 

**Expected behavior**
New values should be retained

strambc, 2021-07-21:
Does this have to do with lag in saving to the portal's backhand?

strambc, 2021-07-21:
Same thing happens when deleting items.
Delete item, save, go back to list, launch same item again, items are still there.
Refresh page --> items are gone.

andreacosolo, 2021-07-21:
When you save, go back to the items list, and open the same item again, it is expected that you do not see your edits immediately, due to indexing. This is true for any type of item in the database.

For this reason, we display a message upon every click on Save: "This Microscope Configuration Item has been updated with the current viewport. This may take a few minutes to take effect." However, this message does not appear in the full screen mode. @utku-ozturk is there any way to do that?

You also describe a (different/related?) issue: edit one component within MicroMeta App and come back to that component, without leaving the App. Here I do see changes immediately (with or without saving), as expected. @strambc could you please add any detail that could help me replicate your issue?

closed 2024-03-25 by strambc