WU-BIMAC/MicroMetaApp-React #398: [4DN-portal] When changing values in a metadata field the value reverts to old value

strambc, 2021-07-22:
**Describe the bug**
Edit value does not stick, it reverts back to old value 

**To Reproduce**
Steps to reproduce the behavior:
1. Edit Microscope
2. Change the value of Name field
3. Value revert to old value (before hitting confirm as the mouse is scrolling towards "Confirm").

andreacosolo, 2021-07-22:
I can reproduce this. In addition, I also noticed automatic tab switching.

**To Reproduce**
1. Edit Microscope. The first tab is Instrument.
2. Click tab InvertedMicroscopeStand.
3. Wait for 5-10 seconds.
4. The selected tab goes back to Instrument.

Both this and the edit value issue described above seem to be unique to the Edit Microscope menu. I cannot reproduce them when I edit another component.

utku-ozturk, 2021-07-27:
I can also reproduce this bug not only in 4DN but also in the electron stand-alone app.

To Reproduce

1. Open stand-alone app (exit full-screen to resize it)
2. Click tab InvertedMicroscopeStand. Edit `Name` field in `General` tab.
3. Resize app
4. The selected tab goes back to Instrument and `Name` field is reverted to the initial state.

strambc, 2021-07-27:
Confirm that this happens on standalone @Akramhar

Akramhar, 2021-08-04:
I think the problem @andreacosolo mentioned it's happening because of the resize hook.
The function hooked to the resize force a re-render of the whole app to constrain the side bar always in the visible viewport.
The re-render obviously force an update of all the views and so all the unsaved data gets flushed.
At the present time I haven't found a better solution to maintain the state and at the same time being able to update the visible viewport.

closed 2022-10-17 by strambc