WU-BIMAC/MicroMetaApp-React #443: [NEW FEATURE] REVAMP APP PRESENTATION UPON LAUNCH --> improve Icon, Initial Icon and About information and presentation

strambc, 2022-07-13:
**Is your feature request related to a problem? Please describe.**

We currently rely uniquely on the REACT features:
- Launch ICON
- Top-left menu About (minimal)
- Also the first screen has no icons
-
**Describe the solution you'd like**

- Custom Micro-Meta App ICON
- Top-left menu About: expanded with credits and license
- Add about information to all windows (under B/W icon)
- Also the first screen has no icons

**Additional context**
@strambc has spoken with Shannon Emsen (4DN-DCIC)  about making a custom launch icon based on the following two mock-ups

[image]
[image]

strambc, 2022-07-13:
About menu item text:

[Launch Icon (top left)]

Micro Meta App is an open, easy-to-use, and powerful software platform that provides an intuitive visual guide to capturing and managing Microscopy Metadata on the basis of the [4DN-BINA extension](https://github.com/WU-BIMAC/NBOMicroscopyMetadataSpecs/tree/master/Model/stable%20version/v02-01) of the [OME data model](https://docs.openmicroscopy.org/ome-model/6.1.1/developers/model-overview.html).

Version X.X.XX-bX
(Model version: X.XX.XX)

(c) Copyright 2018-2023 University of Massachusetts Chan Medical School. All Rights Reserved.
The software is distributed under the terms of the GNU General Public License v3.0.
For more information, visit: https://www.gnu.org/licenses/gpl-3.0.html

strambc, 2022-07-13:
First screen:

[image]

strambc, 2022-07-13:
About text for Screens 1-4 for Manage Instrument and Settings (to be placed below B/W icon when relevant)

Micro-Meta App
(c) Copyright 2018-2023 University of Massachusetts Chan Medical School. All Rights Reserved.

Akramhar, 2022-08-03:
About was done in version 1.7.14, icon is still missing

closed 2025-03-07 by strambc