WU-BIMAC/MicroMetaApp-React #447: [BUG] Reset of field after scroll

Akramhar, 2022-10-12:
**Describe the bug**
Andrea reported that filled fields in the editor are reset after scroll up, scroll down.

**To Reproduce**
Alex and Utku were unable to reproduce the issue.
Caterina testing didn't present the issue.

More investigation and data is required to locate this issue.

**Expected behavior**
Fields should never reset unless Cancel button is pressed

Akramhar, 2022-10-12:
@utku please add the recording of the error here and @andreacosolo if you could please report information about your browser / os setup

utku-ozturk, 2022-10-12:
Screencast: https://gyazo.com/f256d568d5b06b6727fbb4fca682e230

andreacosolo, 2022-10-13:
browser: chrome, Version 104.0.5112.101 (Official Build) (x86_64)

andreacosolo, 2022-10-13:
Utku initially reported the same/a similar issue and was able to resolve it by clearing the browser cache. On my end, I cleared the entire cache, quitted chrome, relaunched. The issue persisted.

strambc, 2022-10-13:
I was able to reproduce it as well. Both on the portal and on Electron 1.7.24-b1. On Electron it required re-sizing the window.

strambc, 2022-10-13:
Is it related to #394?

strambc, 2022-10-13:
NO ERROR associated with the observed behavior on Electron

strambc, 2022-10-17:
Same issue as #398

Akramhar, 2022-10-26:
The issue is related to the resize issue and it present itself, for the best of my knowledge, only in the "edit microscope" and "edit settings" view.
In version 1.7.25-b1 I moved those form to be state, so that the content of the form should be retained after a re render occurs (for example resize event). In the test I performed it seems to correctly persist the form content.
I'll await feedback before marking the issue as closed :)

closed 2024-03-25 by strambc