WU-BIMAC/MicroMetaApp-React #455: [BUG] Error when overing over "Microscope Explorer" area on the top left of MMA-Explorer window

strambc, 2023-05-09:
**Describe the bug**
Error when hovering over "Microscope Explorer" area on the top left of MMA-Explorer window

**To Reproduce**
Steps to reproduce the behavior:
1. Launch MMA
2. Click on 'Continue'
3. Select MMA Explorer
4. hovering over "Microscope Explorer" area
5. ERROR

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Error**
DerivedStateFromProps-begin
2validateModifiers.js:42 Popper: modifier "popoverArrowMargins" provided an invalid "fn" property, expected "function" but got "undefined"
eval @ validateModifiers.js:42
eval @ validateModifiers.js:17
validateModifiers @ validateModifiers.js:13
setOptions @ createPopper.js:106
createPopper @ createPopper.js:235
eval @ usePopper.js:165
invokePassiveEffectCreate @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23487
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3945
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3994
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4056
flushPassiveEffectsImpl @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23574
unstable_runWithPriority @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/scheduler/cjs/scheduler.development.js:468
runWithPriority$1 @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11276
flushPassiveEffects @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23447
(anonymous) @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23324
workLoop @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/scheduler/cjs/scheduler.development.js:417
flushWork @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/scheduler/cjs/scheduler.development.js:390
performWorkUntilDeadline @ /Users/strambc/Dropbox/Workstuff/4DN_Dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/current stable release/2.0.0-b1/2.0.0-b1_micro-meta-app-electron.app/Contents/Resources/app.asar/node_modules/scheduler/cjs/scheduler.development.js:157

strambc, 2023-05-09:
Same error when hovering over "Browse" and "Continue" buttons in the first screen

strambc, 2023-05-09:
Same error when hovering over "MMExplorer" and "MMApp" buttons in the first screen

Akramhar, 2023-05-15:
This comes from search field library, not sure we can fix it