WU-BIMAC/MicroMetaApp-React #462: [BUG] Microscopes List in "Load from local Home Folder" is cut short at 15

strambc, 2023-05-16:
**Describe the bug**
List of Microscopes in "Load from local Home Folder" is cut short at 15.
This is made clear when attempting to open Nikon microscopes: a total of 20 are present but only 15 are displayed.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] MacOS Ventura 13.3.1
 - Version [e.g. 22]  v1.7.25.b1

strambc, 2023-05-16:
This only happens in "Manage Hardware"

strambc, 2024-03-26:
[image]

strambc, 2025-03-06:
Reviewed on 3/6/2025
Still valid
High Priority

Cynthiawzy, 2025-05-29:
UI issue fixed, all of the Manufacturer and Microscope names are properly displayed and user can scroll vertically to have access to all the available selections