WU-BIMAC/MicroMetaApp-React #466: [Save Individual component] MANAGE INSTRUMENT --> add capacity to save individual Hardware component

strambc, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
Creating Microscope Templates is very difficult without the capacity to save and import individual components.

USE CASE: I have described multiple Objectives that I know are commonly used, and I want to save the specifications of each so that I can use it in another microscope or share it with others
IDEAL: Save one at a time + Save All components.

**Describe the solution you'd like**
Add capacity to save individual hardware components

This issue is described in Trello https://trello.com/c/KgEjVI0A

strambc, 2024-03-08:
This is linked to an Electron Issue to Create "Save to home folder" method - MMA_Electron

- Build method to Save to folder in micrometapp/component/categoryName
- Install Save Method to React to allow saving to homefolder on the file system instead of download

Akramhar, 2025-03-18:
This is Cynthia branch that needs to be merged for next release