WU-BIMAC/MicroMetaApp-React #470: [NEW FEATURE] MANAGE INSTRUMENT --> changes to GUI for Saving Components

strambc, 2025-06-24:
- [ ] Change button text: "Component Selector" change to "Functionality Selector"
- [ ] Change tool tip: "Component Selector" button tool tip change to:
- - [ ] Title change to "Functionality Selector"
- - [ ] Text change to "Return to the Micro-Meta App versus Explorer selection window"

- [ ] Change button text: "Home" change to "Micro-Meta App Home"
- [ ] Change tool tip: "Homer" button tool tip change to:
- - [ ] Text change to "Return to the initial window of Micro-Meta App"

Akramhar, 2025-09-08:
point 3 should be fixed in electron 2.1.0 - react 1.8.0

Akramhar, 2025-09-08:
point 4 should be fixed in electron 2.1.0 - react 1.8.0

strambc, 2025-09-17:
Point 1 was only done for Micro-Meta App and not for Micro-Meta Explorer.
Instructions on how fix this are available in Point 5 in issue #476 

Point 2: Title change to "Micro-Meta App Home" was not done

Point 3: DEPRECATED because Home button in MME does not exist.

Point 6: NOT DONE

Akramhar, 2026-01-14:
this should be completed in MMA 1.8.1 and MME 1.1.0